AIAANortheastern/karman-avionics #27: Read and write defines are wrong in wiki

jkleve, 2017-07-08:
Someone should double check me but I'm pretty sure the read is OR'd with 0x01 instead of write. Testing it out on my board, I'm sending 0x01 with my address and the code I get back is SLA+R send, ACK received.

ADKaster, 2017-07-08:
According to this TI doc

http://www.ti.com/lit/an/slva704/slva704.pdf

The eighth bit sent with the address should be R/W'.
That is, if you have your seven bit address, the proper thing to do is shift left 1 then OR with 0x01 for read. So if the wiki says the opposite, it's wrong.

jkleve, 2017-07-09:
Yes, it is wrong. This is what is in the karman-avionics wiki.

```
/* b10001010 == 0x8A */
/* b10001011 == 0x8B */
/*  ^   ^            */
/*  8   B            */
#define DEV_ADDR (0x45) /* This is the address above left shifted 1 */
/* b1000101X --> shift left 1, the last bit is a don't care
   b01000101 --> The rightmost digit drops off 
    ^   ^
    4   5
*/
#define DEV_READ (DEV_ADDR << 1 | 0x00)       /* 0x8A */
#define DEV_WRITE (DEV_ADDR << 1 | 0x01)      /* 0x8B */
```

ADKaster, 2017-07-23:
Should be resolved now

closed 2017-07-28 by ADKaster