AIAANortheastern/karman-avionics #30: Code needs to be fully commented

dschonholtz, 2017-09-17:
I am trying to get the code commented by Monday night so that it is slightly more readable by the time the new members get to look at it.

dschonholtz, 2017-09-17:
I'm gonna start with the drivers. The drivers are going to need to have three things IMO
1) What the driver is actually responsible for doing.
2) What the associated hardware is responsible for doing. 
3) What each function is doing. 

All of that should be in doxy formatted comments I.E. : 

    /**
    * Comments formatted as this I think....
    *
    */

ADKaster, 2017-09-17:
Looking at the reference documentation, the important stuff is a brief and detailed description for each file and function. After that, annotations for structure definitions would be great.

https://www.stack.nl/~dimitri/doxygen/manual/docblocks.html

I believe there's a config option to treat typedefs as structs or something like that so that when you do 

```c
typedef struct
{
      int foo; /**< holds foo for calculating baz */
      float bar; /**< scaling parameter for foo */
} foobar_t;
```
It only sees the type foobar_t and not the anonymous struct in the middle.

dschonholtz, 2017-09-17:
Alright that makes sense. Because I'm a doxygen rookie and I'm brushing off cobwebs for C I am going to start simple with just the file and function names and then I'll go into further documentation afterwards. 
I'll keep updating here as I go with issues accomplishments etc.

dschonholtz, 2017-09-17:
wooooooo alrighty then I have commented code. sorta. Doxygen isn't playing nice and I definitely got more lax about how thorough I wanted commenting to be as time went on. I'm on bus wifi right now and will look through everything again tonight and then probably issue a pull request. But for now I'm not gonna keep looking at it

dschonholtz, 2017-09-18:
About to issue a pull request. Whoever ends up looking it over can send it back to me for more comments if they want. 
Doxygen is not currently working and I by no means have doxygen commenting set up in all of the files. I think that should actually be a separate issue though that I'll open after this.

dschonholtz, 2017-09-21:
I'm gonna go ahead and close this because the original issue is gone now. I can understand the code. 
We also have doxygen that works. Because of that I am closing this ticket. If people can't understand the code and massive chunks need to be commented we might reopen it, but I don't think that is the case.

closed 2017-09-21 by dschonholtz